sensebox/React-Ardublockly #64: Custom MQTT Broker not working

felixerdy, 2021-02-11:
You cannot change the MQTT Broker when using the `Connect to MQTT Broker` block. As you can see in the video below, changes are not applied.

https://user-images.githubusercontent.com/12086004/107620438-f9447d00-6c54-11eb-9832-91bbffee440d.mov

This also applies to the `port`, but `username` and `password` are working

mariopesch, 2021-02-11:
Thanks @felixerdy I'll check it

felixerdy, 2021-02-11:
And, as a follow up, `temperature` in the function `Adafruit_MQTT_Publish` should also be a string when using a custom MQTT Broker I guess. 

[image]

not: `Adafruit_MQTT_Publish temperature = Adafruit_MQTT_Publish(&mqtt, temperature);`
but: `Adafruit_MQTT_Publish temperature = Adafruit_MQTT_Publish(&mqtt, "temperature");`

mariopesch, 2021-02-11:
I'm also thinking about change the whole mqtt thing to the pubsub library instead of the adafruit library.

PR #65 (2021-02-15): fix for custom mqtt server
This PR should close #64
closed 2021-02-15 by mariopesch