sensebox/React-Ardublockly #254: [Blockly][Codegenerator] Add LED Matrix

mariopesch, 2024-01-26:
Add Block and Code for the new LED Matrix.

Might use https://github.com/google/blockly-samples/tree/master/plugins/field-bitmap for the block

mariopesch, 2024-01-29:
See here: https://shop.watterott.com/senseBox-WS2812-LED-Matrix

PaulaScharf, 2024-02-12:
Minimal RGB Matrix Rainbow Example:
```cpp
#include <Adafruit_GFX.h>
#include <Adafruit_NeoMatrix.h>
#include <Adafruit_NeoPixel.h>

#define PIN 2
#define WIDTH 12
#define HEIGHT 8

Adafruit_NeoMatrix RGBMatrix = Adafruit_NeoMatrix(WIDTH, HEIGHT, PIN,
  NEO_MATRIX_TOP     + NEO_MATRIX_LEFT +
  NEO_MATRIX_ROWS + NEO_MATRIX_ZIGZAG,
  NEO_GRB            + NEO_KHZ800);

// created with http://www.rinkydinkelectronics.com/t_imageconverter565.php
const uint16_t bitmap[96] =
{
  0xE940, 0xFB00, 0xFCE0, 0xF703, 0xB7E2, 0x5761, 0x17AA, 0x17B6, 0x15BE, 0x023C, 0x403C, 0x887F, 0xE940, 0xFB00, 0xFCE0, 0xF703,   // 0x0010 (16) pixels
  0xB7E2, 0x5761, 0x17AA, 0x17B6, 0x15BE, 0x023C, 0x403C, 0x887F, 0xE940, 0xFB00, 0xFCE0, 0xF703, 0xB7E2, 0x5761, 0x17AA, 0x17B6,   // 0x0020 (32) pixels
  0x15BE, 0x023C, 0x403C, 0x887F, 0xE940, 0xFB00, 0xFCE0, 0xF703, 0xB7E2, 0x5761, 0x17AA, 0x17B6, 0x15BE, 0x023C, 0x403C, 0x887F,   // 0x0030 (48) pixels
  0xE940, 0xFB00, 0xFCE0, 0xF703, 0xB7E2, 0x5761, 0x17AA, 0x17B6, 0x15BE, 0x023C, 0x403C, 0x887F, 0xE940, 0xFB00, 0xFCE0, 0xF703,   // 0x0040 (64) pixels
  0xB7E2, 0x5761, 0x17AA, 0x17B6, 0x15BE, 0x023C, 0x403C, 0x887F, 0xE940, 0xFB00, 0xFCE0, 0xF703, 0xB7E2, 0x5761, 0x17AA, 0x17B6,   // 0x0050 (80) pixels
  0x15BE, 0x023C, 0x403C, 0x887F, 0xE940, 0xFB00, 0xFCE0, 0xF703, 0xB7E2, 0x5761, 0x17AA, 0x17B6, 0x15BE, 0x023C, 0x403C, 0x887F,   // 0x0060 (96) pixels
};

void setup()
{
  RGBMatrix.setBrightness(15);
  RGBMatrix.begin();
  RGBMatrix.drawRGBBitmap(0,0, bitmap, WIDTH, HEIGHT);
  RGBMatrix.show();
}

void loop()
{

}

```

mariopesch, 2024-02-12:
@Thiemann96 I just started the integration as we now have a matrix here and can also test the stuff

PaulaScharf, 2024-02-12:
Minimal scrolling text from sensor (ToF) example:
```cpp
#include <Wire.h>
#include <vl53l8cx_class.h>

#include <Adafruit_GFX.h>
#include <Adafruit_NeoMatrix.h>
#include <Adafruit_NeoPixel.h>

#define PIN 2

Adafruit_NeoMatrix RGBMatrix = Adafruit_NeoMatrix(12, 8, PIN,
  NEO_MATRIX_TOP     + NEO_MATRIX_LEFT +
  NEO_MATRIX_ROWS + NEO_MATRIX_ZIGZAG,
  NEO_GRB            + NEO_KHZ800);

VL53L8CX sensor_vl53l8cx_top(&Wire, -1, -1);
String dataStr = "";

void setup()
{
  Wire.begin();
  Wire.setClock(1000000); //Sensor has max I2C freq of 1MHz
  sensor_vl53l8cx_top.begin();
  sensor_vl53l8cx_top.init_sensor();
  sensor_vl53l8cx_top.vl53l8cx_set_ranging_frequency_hz(30);
  sensor_vl53l8cx_top.vl53l8cx_set_resolution(VL53L8CX_RESOLUTION_8X8);
  sensor_vl53l8cx_top.vl53l8cx_start_ranging();

  RGBMatrix.setBrightness(15);
  RGBMatrix.setTextWrap(false);
  RGBMatrix.setTextColor(RGBMatrix.Color(255, 0, 0));
  RGBMatrix.setTextSize(1);
  RGBMatrix.begin();
}

void loop()
{
    float min = getVl53l8cxMin();
    RGBMatrix.setTextColor(getLedColorHSV(map(min,0,2000,10,310),1,1));
    char minChar[20];
    int length = snprintf(minChar, sizeof(minChar), "%.1f", min);
    for(int i = 0; i<length*6-4; i++) {
      RGBMatrix.fillScreen(0);
      RGBMatrix.setCursor(6-i, 0);
      RGBMatrix.print(minChar);
      RGBMatrix.show();
      delay(100);
    }
    RGBMatrix.fillScreen(0);
    RGBMatrix.show();
    delay(100);
}

float oldVl53l8cxMin = -1.0;
float getVl53l8cxMin() {
  VL53L8CX_ResultsData Results;
  uint8_t NewDataReady = 0;
  uint8_t status;

  status = sensor_vl53l8cx_top.vl53l8cx_check_data_ready(&NewDataReady);

  if ((!status) && (NewDataReady != 0)) {
    sensor_vl53l8cx_top.vl53l8cx_get_ranging_data(&Results);
    float min = 10000.0;
    for(int i = 0; i < VL53L8CX_RESOLUTION_8X8*VL53L8CX_NB_TARGET_PER_ZONE; i++) {
      if((&Results)->target_status[i]!=255){
        float distance = (&Results)->distance_mm[i];
        if(min > distance) {
          min = distance;
        }
      }
    }
    oldVl53l8cxMin = (min==10000.0) ? 0.0 : min;
  }
  return oldVl53l8cxMin;
}

uint16_t getLedColorHSV(int h, double s, double v) {
  //this is the algorithm to convert from RGB to HSV
  double r=0; 
  double g=0; 
  double b=0;

  double hf=h/60.0;

  int i=(int)floor(h/60.0);
  double f = h/60.0 - i;
  double pv = v * (1 - s);
  double qv = v * (1 - s*f);
  double tv = v * (1 - s * (1 - f));

  switch (i)
  {
  case 0: //rojo dominante
    r = v;
    g = tv;
    b = pv;
    break;
  case 1: //verde
    r = qv;
    g = v;
    b = pv;
    break;
  case 2: 
    r = pv;
    g = v;
    b = tv;
    break;
  case 3: //azul
    r = pv;
    g = qv;
    b = v;
    break;
  case 4:
    r = tv;
    g = pv;
    b = v;
    break;
  case 5: //rojo
    r = v;
    g = pv;
    b = qv;
    break;
  }

  //set each component to a integer value between 0 and 255
  int red=constrain((int)255*r,0,255);
  int green=constrain((int)255*g,0,255);
  int blue=constrain((int)255*b,0,255);

  return RGBMatrix.Color(red, green, blue);
}
```

commit 236f4248 (2024-03-18): add blocks for the led matrix

closes #254
PR #261 (2024-04-09): Development
This PR 
closes #255 
closes #258 
closes #249 
closes #259 
closes #273 
closes #281 
closes #254 
closes #247 
closes #275
closes #248
closed 2024-04-09 by mariopesch