sensebox/React-Ardublockly #297: [Blockly][UI] adjust image of sensebox while compiling

PaulaScharf, 2024-04-03:
[image]

The image should depend on the chosen sensebox model, just like the background image

BjoernLuig, 2024-11-19:
I suggest, a small white Window (for better contrast) which stays open after the download until closed. It could show

- the description of the down and upload process depend on the selected board type like @PaulaScharf and @verena-witte  suggested
- additional links to information like resetting the MCU-S2 after using micropython (see https://docs.sensebox.de/docs/misc/circuitpython_esp32/)
- the better error messages done by @Thiemann96 if the download failed.

My experience with students who are often new to blockly and senseBox suggests, that blockly needs a clearer description of what is going on and expected from the user to do after the download. As I am not a real web dev i will just leave this here for now.

Thiemann96, 2025-01-27:
added this to the release in 2 weeks. @georgi-s is taking a look at this!