sensebox/React-Ardublockly #316: [Blockly][Codegenerator] Water temperature sensor

PaulaScharf, 2024-06-26:
Water temperature sensor doesnt return its value.

Following blockly code:
[image]

results in following arduino code:
```
#include <senseBoxIO.h>
#include <OneWire.h> // http://librarymanager/All#OneWire
#include <DallasTemperature.h> // http://librarymanager/All#DallasTemperature

#define ONE_WIRE_BUS 1
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

float getWaterTemp(){
sensors.requestTemperatures();
sensors.getTempCByIndex(0);
}

void setup() {
SerialUSB.begin(300);
sensors.begin();

}

void loop() {
SerialUSB.println(getWaterTemp());
delay(1000);

}
```

I think `getWaterTemp` needs to return sth or Im completely misunderstanding this code.

Also potentially for the MCU-S2 the pin needs to change from 1 to 3 (and 3 to 5 and 5 to 7)

mariopesch, 2024-07-05:
Extension of this issue:

Update the block and allow reading multiple temperature sensors attached to one onewire adapter board. It's possible to hook up to 6 Sensors to one board.

PaulaScharf, 2024-09-13:
dfrobot sensor doesnt work on old MCU because analogReadMilliVolts is used instead of analogRead

PR #327 (2024-09-16): Development
New development branch with various fixes listed below

**WORK IN PROGRESS** 

- fixes #316 
- closes #325 
- closes #324
- closes #307 
- closes #311 
- closes #330
closed 2024-09-16 by Thiemann96