sensebox/React-Ardublockly #320: Fehlende/ falsche Verlinkung im Hilfe-Kasten

verena-witte, 2024-07-17:
Einige Verlinkungen im Hilfe-Kasten führen zu einem Error: https://blockly.sensebox.de/null

Vorschlag: Im Hilfekasten direkt die jeweilige Lernkarte einblenden (+ Link zum Sensor-Wiki, falls vorhanden)

Thiemann96, 2024-07-18:
Hier wäre es gut zu wissen, welche bzw. bei welchen Blöcken ?

verena-witte, 2024-07-22:
Also bei einigen Blöcken (z.B. Helligkeit- und UV) kommt bei dem Button 'Sensor-Informationen' 'no data available'. Bei anderen Blöcken (z.B. hier https://docs.sensebox.de/hardware/sensors/feinstaub-sds011) ist die Verlinkung zum Sensor-Wiki falsch. Brauchst du wirklich eine Liste der Blöcke (das werden sehr viele sein) oder kannst du das direkt übers Sensor-Wiki herausfinden, welche Seiten da (nicht) funktionieren?

mariopesch, 2024-07-22:
sollen wir die Informationen überhaupt noch aus dem Sensor Wiki ziehen? Da wir nur die Infos zu den Sensoren und nicht zu den ganzen anderen Blöcken aus dem Sensor wiki bekommen würde ich fast vorschlagen, dass wir keine Infos mehr aus dem Sensor wiki holen

closed 2025-01-10 by Thiemann96