sensebox/React-Ardublockly #333: MCU-S2 can't plot diagram on display due to library error

eduardocs21, 2024-09-03:
**Describe the bug**
A blockly code (example attached) to plot a diagram on the OLED-Display, showing the values the sensor is measuring, leads to a compiling error on the senseBox MCU-S2:

"exit":"Command failed: 1: Uncaught Fatal Exception","process":"/root/Arduino/libraries/OLED_Plot_Library/src/Plot.cpp:2:10: fatal error: avr/dtostrf.h: No such file or directory\\n #include <avr/dtostrf.h>\\n ^\~~\\ncompilation terminated.\\nError during build: exit status 1\\n"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://blockly.sensebox.de/
2. Put together a program using the block "plot diagram", example given with the screenshot below (xml-files cant be uploaded here)
3. Compile the program using the orange button.
4. See error

**Expected behavior**
The code should compile without error. Then, when transferring it onto the senseBox MCU-S2, it should plot the diagram as intended.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Firefox
 - Version: 129.0.2 (64-Bit)

**Additional context**
The diagram works on the senseBox MCU (S1) without issues.

Thiemann96, 2024-09-12:
I updated the library, changes should soon be online

Thiemann96, 2024-09-16:
fixed by https://github.com/sensebox/SSD1306-Plot-Library/pull/1

closed 2024-09-16 by Thiemann96