sensebox/React-Ardublockly #334: [Blockly][Codegenerator] openSenseMap upload via WiFi generates compiling errors on booth MCU and MCU-S2

BjoernLuig, 2024-09-05:
### Block description and Expected block behaviour 
The following sketch is meant to to upload a single value onto the openSenseMap via Wifi
[image]

### Error
On the MCU I recieve the complier error
`{"exit":"Command failed: 1: Uncaught Fatal Exception","process":"/tmp/1b4411a0889d67369ce9f36dc56d32b4/sketch/sketch.ino:6:10: fatal error: WiFi.h: No such file or directory\n #include <WiFi.h>\n ^~~~~~~~\ncompilation terminated.\nError during build: exit status 1\n"}`

Om the MCU-S2 I recieve the compiler error
`{"exit":"Command failed: 1: Uncaught Fatal Exception","process":"/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino: In function 'void writeMeasurementsToClient()':\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:72:7: error: 'client' was not declared in this scope\n client.print(buffer);\n ^~~~~~\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:72:7: note: suggested alternative: 'Client'\n client.print(buffer);\n ^~~~~~\n Client\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino: In function 'void submitValues()':\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:80:7: error: 'client' was not declared in this scope\n if (client.connected()) {\n ^~~~~~\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:80:7: note: suggested alternative: 'Client'\n if (client.connected()) {\n ^~~~~~\n Client\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:89:17: error: 'client' was not declared in this scope\n connected = client.connect(_server, 0);\n ^~~~~~\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:89:17: note: suggested alternative: 'Client'\n connected = client.connect(_server, 0);\n ^~~~~~\n Client\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:130:2: error: 'NVIC_SystemReset' was not declared in this scope\n NVIC_SystemReset();\n ^~~~~~~~~~~~~~~~\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino: In function 'void loop()':\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:170:100: error: conversion from 'int' to 'String' is ambiguous\n printOnDisplay(\"WiFi-Status\", String(WiFi.status()), 0, \"Batterie\", String(barrary_voltage), \"V\");\n ^\nIn file included from /root/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/Arduino.h:188,\n from /tmp/cbde14f3f85bf58b944b2bb2094d131b/sketch/sketch.ino.cpp:1:\n/root/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/WString.h:65:9: note: candidate: 'String::String(const __FlashStringHelper*)'\n String(const __FlashStringHelper *str) : String(reinterpret_cast<const char*>(str)) {}\n ^~~~~~\n/root/.arduino15/packages/esp32/hardware/esp32/2.0.16/cores/esp32/WString.h:59:9: note: candidate: 'String::String(const char*)'\n String(const char *cstr = \"\");\n ^~~~~~\n/tmp/2232684d68655f045f06bd1f19d1ba1c/sketch/sketch.ino:34:10: note: initializing argument 3 of 'void printOnDisplay(String, String, String, String, String, String)'\n void printOnDisplay(String title1, String measurement1, String unit1, String title2, String measurement2, String unit2) {\n ^~~~~~~~~~~~~~\n\nError during build: exit status 1\n"}`

Thiemann96, 2024-09-12:
When building the Blocks like shown I get no errors. Probably the blocks were set with a different Board and afterwards the board was changed, no?

BjoernLuig, 2024-11-19:
That's the problem. I changed the title for more clearance.