sensebox/React-Ardublockly #369: Implement Classroom Mode

mariopesch, 2024-11-25:
Implement a "Classroom Mode" in senseBox Blockly, inspired by Tinkercad's classroom features, to streamline teaching and collaborative learning in educational environments. The Classroom Mode should support educators in managing and guiding multiple students working on Blockly projects, enabling a structured and interactive learning experience.

Key Features:

**Teacher Dashboard:**

- View and manage student projects
- create classrooms and add students via name and nickname

**Student Accounts:**

- Enable students to log in with a classroom Code and the given nickname
- allow students to save and share their projects within the classroom
- Facilitate sharing of block code between students and the teacher.
- Allow teachers to broadcast example code or templates for students to follow.

User Authentication: Teachers/Classroom creators should login with their openSenseLab Account or use google oAuth

mariopesch, 2024-11-25:
dev branch: https://github.com/sensebox/React-Ardublockly/tree/feat/addClassroom

Thiemann96, 2025-01-28:
will move this into the backlog for now, lets try to have it ready for the release in march