sensebox/React-Ardublockly #381: Multiple Tooltips in Math Blocks do not work

BjoernLuig, 2024-12-23:
### Multiple Tooltips in Math Blocks do not work

It seems to be a Problem with the setTooltips() function, wich theoretically works with a function as parameter, which does throw a react-markdown warning and does not give a tooltip. This Problem persists over multiple math blocks.

I am currently fixing two other issues within the math blocks occurring with the new blockly version in the development branch. If someone has a short solution, I could do all in one. Otherwise, I would make the tooltips constant for now.

[image]
[image]
[image]

mariopesch, 2024-12-23:
have you check if `var thisBlock = this` is returning the current block?

BjoernLuig, 2024-12-30:
thisBlock refereed to the correct block, but the setTooltip function coud not handle any function as parameter, only strings.

I found a workaround by updateing the tooltip through the generators. See [382](https://github.com/sensebox/React-Ardublockly/pull/382). This was thoroughly tested for all changed math blocks.

BjoernLuig, 2025-01-06:
I will close this issue as it works with this workaround for know
https://github.com/sensebox/React-Ardublockly/pull/382

mariopesch, 2025-01-06:
lets keep it open until it is merged into the main.

closes via #382

closed 2025-02-14 by Thiemann96