sensebox/React-Ardublockly #401: create function block buggy

PaulaScharf, 2025-02-18:
**Describe the bug**
function block buggy

**To Reproduce**
Take a Create-Function Block. Then try to take another function block. An error will appear in the console and the positioning of the function blocks will fail. This makes implementing functions in blockly basically impossible at the moment.

**Screenshots**

[image]

**Desktop (please complete the following information):**

- OS: Ubuntu
- Browser: Firefox

Thiemann96, 2025-02-20:
weird. didnt make any changes to that block AFAIK. Maybe because of the blockly update?

Thiemann96, 2025-02-20:
While testing I found a workaraound: Copy and paste the block instead of dragging it from the toolbox

Thiemann96, 2025-02-20:
Also using the backpack works. 

Have not been able to find anything in the Blockly repo and I can't seem to make sense of the error you posted ... 🤔

mariopesch, 2025-02-20:
Yes it is probably an issue which is connected to the blockly update. In the Blockly update some mutator (the small gear functionality) functions got reworked and might be complete deprecated.

The procedures definitly had some issues even before the the blockly update, as the implementation of the types was not perfect.

closed 2025-02-21 by Thiemann96