sensebox/React-Ardublockly #463: Issues with BME680 Block and MCU S2

kgalb01, 2025-04-29:
When trying to create a sketch for the MCU S2, it won’t compile if it includes anything related to the BME680 sensor.
The issue seems to be that the library used was updated and now requires a built-in LED (see screenshot).

The block still works fine with the MCU S1, though.

I created a sketch that uses the Adafruit library instead, which works with the MCU S2. I can't attach it here though but I used the library "#include <Adafruit_BME680.h>".

[image]

mariopesch, 2025-04-29:
@kgalb01 the adafruit library is not as advanced as the original bosch bsec library. 

We can simply change the error code from the the builtin LED to the RGB LED or remove it for the MCU-S2

kgalb01, 2025-04-29:
Sorry, I closed the issue by accident. I’ll try to get the code working with the library you suggested and report back.

mariopesch, 2025-05-16:
@kgalb01 any update on this?

kgalb01, 2025-05-16:
Not yet, I had different priorities this week. I think I'll get around this issue next week

PaulaScharf, 2025-09-22:
I ran into this issue while checking out senseBoxOS. Here is a sketch:
```cpp
#include <bsec2.h>

float bmeTemperatur;
float bmeHumidity;
double bmePressure;
float bmeIAQ;
float bmeIAQAccuracy;
int bmeCO2;
float bmeBreathVocEquivalent;

Bsec2 bme680;

bool BME680begin() {
  bme680.begin(BME68X_I2C_ADDR_LOW, Wire);
  if (bme680.status != BSEC_OK || bme680.sensor.status != BME68X_OK) {
    return false;
  }

  bsecSensor sensorList[14] = {
    BSEC_OUTPUT_IAQ,
    BSEC_OUTPUT_RAW_TEMPERATURE,
    BSEC_OUTPUT_RAW_PRESSURE,
    BSEC_OUTPUT_RAW_HUMIDITY,
    BSEC_OUTPUT_RAW_GAS,
    BSEC_OUTPUT_STABILIZATION_STATUS,
    BSEC_OUTPUT_RUN_IN_STATUS,
    BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_TEMPERATURE,
    BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_HUMIDITY,
    BSEC_OUTPUT_STATIC_IAQ,
    BSEC_OUTPUT_CO2_EQUIVALENT,
    BSEC_OUTPUT_BREATH_VOC_EQUIVALENT,
    BSEC_OUTPUT_GAS_PERCENTAGE,
    BSEC_OUTPUT_COMPENSATED_GAS
  };

  bme680.updateSubscription(sensorList, ARRAY_LEN(sensorList), 1); // 1Hz
  if (bme680.status != BSEC_OK || bme680.sensor.status != BME68X_OK) {
    return false;
  }
  return true;
}

void setup() {
  Wire.begin();

  Serial.begin(115200);
  BME680begin();
}

void loop() {

  if (bme680.run()) {
    bmeTemperatur = bme680.getData(BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_TEMPERATURE).signal;
    bmeHumidity = bme680.getData(BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_HUMIDITY).signal;
    bmePressure = bme680.getData(BSEC_OUTPUT_RAW_PRESSURE).signal;
    bmeIAQ = bme680.getData(BSEC_OUTPUT_IAQ).signal;
    bmeIAQAccuracy = bme680.getData(BSEC_OUTPUT_STATIC_IAQ).signal;
    bmeCO2 = bme680.getData(BSEC_OUTPUT_CO2_EQUIVALENT).signal;
    bmeBreathVocEquivalent = bme680.getData(BSEC_OUTPUT_BREATH_VOC_EQUIVALENT).signal;
  }

  Serial.println(bmeTemperatur);
  delay(1000);

}
```
I used the newer bsec library ([bsec2](https://github.com/boschsensortec/Bosch-BSEC2-Library)) because the old one seems to not be maintained anymore.